Xinyuan-LilyGO/T-Encoder-Pro #2: Display physical stability

nikthefix, 2024-06-12:
Received this unit today. 
Display looks fantastic!
Love the form factor.
On my unit the display glass is able to rotate +/- a degree or so with respect to the main board housing when turning the encoder.
I'm concerned that this can't be good for the FFC connection to the main board.
I imagine it's just tolerances in the sub-assemblies.

Could you upload a step model of the internals?

Many thanks for another terrific product.
And I'm really looking forward to the T-Circle-S3

nik

awilley, 2024-07-04:
I agree about the encoder ring. Really needs to be larger. Will still look great.

nikthefix, 2024-07-04:
@awilley 
I'm temporarily using the foam insert ring which came in the packaging. It's about the right size for me. 
I'm gonna try to construct a contoured brushed aluminium ring - like a mini Nest thermostat.
I think that for this sort of product, an absolute magnetic encoder with no detents would be better. The 'detents' could then be haptics from the mini speaker - nicer for a dial with an undefined application. If the encoder was inductive then the ring hardware could be fabricated from traces on pcb material - no magnets or hall sensors, just a small cheap inductive positional sensor (LX3302A).

awilley, 2024-07-04:
@nikthefix lol, I also kept the foam ring on mine. Shoot me a picture of the ring when you get it built. 
I agree about the detents. I was thinking of using a compact haptic motor like 1597-1244-ND (part id) on digikey or similar. Didn't think of just using the speaker, but that's clever and might work well enough depending on the speaker.
That inductive sensor looks really interesting for this. 
I think this has to be tried. 
Are you planning on putting something together?
I was at one point planning to build a more compact (in height) version of this for some of my small project applications I have.

nikthefix, 2024-07-06:
@awilley Yes I'm preparing an inductive ring design for the 1.43" amoled when it is launched. I only have lilgo drawings at this point but it will be the perfect test bed and saves me having to make a custom board. I already experimented with a linear scale inductive encoder using the Microchip ic. It's a fantastic solution which really works and can be made cheaply to just about any 2D form factor. The esp has to decode the relationship between 2 continuous sinewaves - which it can do with no sweat!

Llgok, 2025-01-02:
Closed due to prolonged inactivity.

closed 2025-01-02 by Llgok