Xinyuan-LilyGO/T-Encoder-Pro #10: Arduino instructions need updating

eslindsey, 2024-08-01:
### IDE and version or firmware

Arduino IDE 2.3.2

### Operating System

Windows 10

### Description

# Setup

- Uninstalled other boards and libraries.

Performed the steps indicated in the "Arduino" section of the instructions:

- Installed boards esp32 by Espressif Systems 2.0.14 (so this is not related to issue #6 or #7)
- Selected DFRobot Firebeetle 2 ESP32-S3 based on platformio.ini
- Copied library files from the repository to my library directory
- Adjusted the settings in the Tools menu to match those provided, except Board since it contradicts platformio.ini

# Expected Behavior

Example program running on the device.

# Actual Behavior

With Board set according to platformio.ini, compilation succeeds but upload fails:

```
Sketch uses 1883781 bytes (143%) of program storage space. Maximum is 1310720 bytes.
Global variables use 95812 bytes (29%) of dynamic memory, leaving 231868 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

With Board set to Adafruit Feather ESP32-S3 No PSRAM, upload is successful.

# Resolution

Instructions should not indicate to choose the board based on platformio.ini. Or, the board in platformio.ini should match the board required for a successful upload (Adafruit Feather ESP32-S3 No PSRAM, per the README).

### Sketch

```cpp
Unmodified code from ./examples/Lvgl_CIT/
```

### Debug Message

```plain
Sketch uses 1898745 bytes (90%) of program storage space. Maximum is 2097152 bytes.
Global variables use 107508 bytes (32%) of dynamic memory, leaving 220172 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM10
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 64:e8:33:4d:13:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001dffff...
Flash will be erased from 0x00410000 to 0x00438fff...
Compressed 22624 bytes to 14245...
Writing at 0x00000000... (100 %)
Wrote 22624 bytes (14245 compressed) at 0x00000000 in 0.4 seconds (effective 425.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 136...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (136 compressed) at 0x00008000 in 0.1 seconds (effective 411.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 520.8 kbit/s)...
Hash of data verified.
Compressed 1899104 bytes to 1071477...
Writing at 0x00010000... (1 %)
Writing at 0x0001ba85... (3 %)
Writing at 0x00024fd8... (4 %)
Writing at 0x00030bf8... (6 %)
Writing at 0x000414ce... (7 %)
Writing at 0x00056c66... (9 %)
Writing at 0x00060414... (10 %)
Writing at 0x00066200... (12 %)
Writing at 0x0006c9c6... (13 %)
Writing at 0x00072b2f... (15 %)
Writing at 0x00078879... (16 %)
Writing at 0x0007e1f3... (18 %)
Writing at 0x00083c29... (19 %)
Writing at 0x00089821... (21 %)
Writing at 0x0008f789... (22 %)
Writing at 0x00095130... (24 %)
Writing at 0x0009aa6c... (25 %)
Writing at 0x000a06a7... (27 %)
Writing at 0x000a65b4... (28 %)
Writing at 0x000b363d... (30 %)
Writing at 0x000bda3f... (31 %)
Writing at 0x000c5407... (33 %)
Writing at 0x000cbddc... (34 %)
Writing at 0x000d219e... (36 %)
Writing at 0x000d8877... (37 %)
Writing at 0x000e09c8... (39 %)
Writing at 0x000ee1d5... (40 %)
Writing at 0x000f7314... (42 %)
Writing at 0x00102f19... (43 %)
Writing at 0x00111825... (45 %)
Writing at 0x0011e54d... (46 %)
Writing at 0x00123ff2... (48 %)
Writing at 0x001295b4... (50 %)
Writing at 0x0012e6db... (51 %)
Writing at 0x001335f0... (53 %)
Writing at 0x00138f72... (54 %)
Writing at 0x0013df12... (56 %)
Writing at 0x00143146... (57 %)
Writing at 0x00148270... (59 %)
Writing at 0x0014da04... (60 %)
Writing at 0x0015357a... (62 %)
Writing at 0x00158baa... (63 %)
Writing at 0x0015e29c... (65 %)
Writing at 0x00163897... (66 %)
Writing at 0x001688d4... (68 %)
Writing at 0x0016d69b... (69 %)
Writing at 0x00172636... (71 %)
Writing at 0x001774fd... (72 %)
Writing at 0x0017c708... (74 %)
Writing at 0x00181988... (75 %)
Writing at 0x001868ff... (77 %)
Writing at 0x0018c30a... (78 %)
Writing at 0x001916fa... (80 %)
Writing at 0x0019658b... (81 %)
Writing at 0x0019b669... (83 %)
Writing at 0x001a0751... (84 %)
Writing at 0x001a5a11... (86 %)
Writing at 0x001aafa1... (87 %)
Writing at 0x001b0d59... (89 %)
Writing at 0x001b623c... (90 %)
Writing at 0x001bb846... (92 %)
Writing at 0x001c3ec2... (93 %)
Writing at 0x001cc665... (95 %)
Writing at 0x001d1dfb... (96 %)
Writing at 0x001d76fe... (98 %)
Writing at 0x001dd4c5... (100 %)
Wrote 1899104 bytes (1071477 compressed) at 0x00010000 in 17.2 seconds (effective 883.7 kbit/s)...
Hash of data verified.
Compressed 165536 bytes to 106746...
Writing at 0x00410000... (14 %)
Writing at 0x00418667... (28 %)
Writing at 0x0041dbce... (42 %)
Writing at 0x0042459a... (57 %)
Writing at 0x0042a572... (71 %)
Writing at 0x0042feab... (85 %)
Writing at 0x004352c0... (100 %)
Wrote 165536 bytes (106746 compressed) at 0x00410000 in 1.7 seconds (effective 778.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
```

Llgok, 2024-08-19:
@eslindsey 
To use the DFRobot Firebeetle 2 ESP32-S3 development board in the Arduino IDE, you can add the following configurations:
[image]
[image]

Llgok, 2024-09-05:
Closed due to prolonged lack of response

closed 2024-09-05 by Llgok