Xinyuan-LilyGO/T-Encoder-Pro #12: Upload Failure No serial data received.

jhordies, 2024-08-15:
### IDE and version or firmware

Platform.io

### Operating System

Win10, OSX

### Description

I can connect to the serial port and get some debug traces from ie: Wifi-Test.
But when I try to updload, from Platform.io on both Mac and Windows and even from the Flashing tool, the device restarts, doesn't beep and stays on a black screen.

### Sketch

```cpp
I used the latest version of this repo
```

### Debug Message

```plain
Uploading .pio/build/Rotary_Encoder/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem14201
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 64:e8:33:4d:25:68
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600

A fatal error occurred: No serial data received.
```

### Other steps to Reproduce

Both Mac and Windows and even from the Flashing tool, the device restarts, doesn't beep and stays on a black screen.
I also tried and installed the VCP drivers on both Mac & Windows https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads without success.
Holding the reset button while plugging it in doesn't help either.

Am I stuck with the demo firmware ?

Llgok, 2024-08-16:
@jhordies 
It seems that the computer is not recognizing the USB port of the device. You can try pressing the BOOT button before powering on the device to see if the computer's device manager recognizes a port instead of pressing the reset button. The BOOT button is located inside the middle position of the board's casing and cannot be seen from the outside. You can place the board flat on a table, and pressing the button with a trigger feeling on the screen indicates that you have pressed the BOOT button.
[image]

jhordies, 2024-08-16:
Thank you!
It worked from the first try, keep the device squeezed while connecting the USB port, and the upload worked flawlessly.I kept it pressed (might not be needed) during the all process. 
I think this deserves to be in the readme as I spent quite some time in this issue without guessing the right button to press. It could be useful to other noobs

closed 2024-08-16 by jhordies