Xinyuan-LilyGO/T-Encoder-Pro #20: pin_config.h missing

rumoldus, 2025-08-13:
### Operating System

Windows 11 64 bit Intel

### IDE

Arduino 2.3.6

### Software Package Version

ESP32S3 DEV Module from Espressif ESP boards 3.3.0 by Espressif

### Description

Trying to run the most simple example of rotary_encoder and it fails on the 2nd line. "include "pin_config.h". This file does not exist in the whole T_encoder pro folder structure. There is also no instruction in the INO file to create this file or where to find it. Very disappointing that the most simple example cannot be run.

### Sketch

```cpp
#include <Arduino.h>
#include "pin_config.h"
```

### Debug Message

```plain
...\T-Encoder-Pro-arduino-esp32-libs_V3.0.7\examples\Rotary_Encoder\Rotary_Encoder.ino:9:10: fatal error: pin_config.h: No such file or directory
    9 | #include "pin_config.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: pin_config.h: No such file or directory
```

Llgok, 2025-08-14:
@rumoldus 
The pin_config.h file is located in the [./libraries/Mylibrary/pin_config.h directory](https://github.com/Xinyuan-LilyGO/T-Encoder-Pro/blob/arduino-esp32-libs_V3.0.7/libraries/Mylibrary/pin_config.h). You need to manually move it to the Arduino library directory.

For the moving operation, please refer to [QuickStart](https://github.com/Xinyuan-LilyGO/T-Encoder-Pro?tab=readme-ov-file#arduino).

rumoldus, 2025-08-14:
Many thanks for the very quick reaction.

closed 2025-08-15 by Llgok